parthkharade/ECEN-5713-FinalProject #1: Buildroot Environment Bringup for BeagleBone Black

parthkharade, 2024-04-10:


pranjalgupta1901, 2024-04-15:
# Buildroot Environment Bringup of BeagleBone Black
## Blockers 
- UART Setup
- Ethernet Setup in order to do SSH
## Work Done to solve the blockers
- I started with building image for BeagleBone Black using Buildroot, setting its UART for serial terminal and SSH using ethernet. The Image was successfully built and the board was booted everytime using SD Card.
- I was able to see that ethernet was getting detected and its driver is also working fine. I was not able to do ssh even though I was connected with the internet and I was able to send messages but was able to recieve it as the messages were getting lost somewhere. I found out that there is a hardware problem associated with the Ethernet I had in my board which is LAN 8710. Therefore, to proceed with the project, I went ahead with building the image for Raspberry Pi 3B.

# Buildroot Environment Bringup of Raspberry Pi 3B
- Raspberry Pi successfully booting up with SD Card with SSH support with the help of Buildrooot Image
- Blockers : No Blockers during the process

pranjalgupta1901, 2024-04-15:
# Buildroot Environment Bringup of BeagleBone Black

## Blockers
-  UART Setup
-  Ethernet Setup in order to do SSH

## Work done
 -  I started with building image for BeagleBone Black using Buildroot, setting its UART for serial terminal and SSH using ethernet. The Image was successfully built and the board was booted everytime using SD Card.
- I was able to see that ethernet was getting detected and its driver is also working fine. I was not able to do ssh even though I was connected with the internet and I was able to send messages but was able to recieve it as the messages were getting lost somewhere. I found out that there is a hardware problem associated with the Ethernet I had in my board which is LAN 8710. This issue is also addressed by BeagleBone officially whose reference can be found this [link.](https://github.com/beagleboard/beaglebone-black/issues/4). Therefore, to proceed with the project and due to non availability of a different board, I went ahead with building the image for Raspberry Pi 3B.

# Buildroot Environment Bringup of Raspberry Pi 3B
## Blockers
- None
## Work Done
- Raspberry Pi successfully booting up with SD Card with SSH support using Ethernet with the help of Buildroot Image

closed 2024-04-15 by parthkharade